Jeff-Ciesielski/Boms-Away #3: Can't make it work in debian jessie

marcos-paltatech, 2016-03-16:
Hi,
maybe you could help me debug this, this is the log:

`marcos@bluestreak:~/electronica/tools/kicad_ext_tools/Boms-Away$ python bomsaway.py             
[INFO   ] [Logger      ] Record log in /home/marcos/.kivy/logs/kivy_16-03-16_9.txt
[INFO   ] [Kivy        ] v1.9.1
[INFO   ] [Python      ] v2.7.9 (default, Mar  1 2015, 12:57:24) 
[GCC 4.9.2]
[INFO   ] [Factory     ] 179 symbols loaded
[INFO   ] [Image       ] Providers: img_tex, img_dds, img_gif, img_pygame, img_pil (img_ffpyplayer ignored)
[INFO   ] [Text        ] Provider: pygame
[INFO   ] [Window      ] Provider: pygame(['window_egl_rpi'] ignored)
[INFO   ] [GL          ] OpenGL version <3.0 Mesa 10.3.2>
[INFO   ] [GL          ] OpenGL vendor <Intel Open Source Technology Center>
[INFO   ] [GL          ] OpenGL renderer <Mesa DRI Intel(R) Sandybridge Mobile >
[INFO   ] [GL          ] OpenGL parsed version: 3, 0
[INFO   ] [GL          ] Shading version <1.30>
[INFO   ] [GL          ] Texture max size <8192>
[INFO   ] [GL          ] Texture max units <16>
[INFO   ] [Window      ] virtual keyboard not allowed, single mode, not docked
 Traceback (most recent call last):
   File "bomsaway.py", line 537, in <module>
     BomsAwayApp().run()
   File "/usr/local/lib/python2.7/dist-packages/kivy/app.py", line 802, in run
     root = self.build()
   File "bomsaway.py", line 453, in build
     self.ds = datastore.Datastore()
 AttributeError: 'module' object has no attribute 'Datastore'
`

I am missing something?
sqlalchemy and navigationdrawer are already installed...

Jeff-Ciesielski, 2016-03-17:
That's odd, looks like an import conflict.

Can you post the result of 'pip freeze'?

marcos-paltatech, 2016-03-17:
Sure, here it is

$ pip freeze
asteval==0.9.1
astroid==1.2.1
beautifulsoup4==4.4.1
bzr==2.7.0.dev1
BzrTools==2.6.0
cffi==0.8.6
characteristic==0.1.0
chardet==2.3.0
Cheetah==2.4.4
CherryPy==3.6.0
configobj==5.0.6
cov-core==1.14.0
coverage==3.7.1
cryptography==0.6.1
Cython==0.23
datastore==0.3.6
DBUtils==1.1
decorator==3.4.0
defusedxml==0.4.1
deluge==1.3.10
docutils==0.12
execnet==1.2.0
Flask==0.10.1
future==0.15.2
gpxpy==0.9.8
gyp==0.1
html5lib==0.999
httplib2==0.9
itsdangerous==0.24
Jinja2==2.7.3
keyring==4.0
kicost==0.1.10
Kivy==1.9.1
Kivy-Garden==0.1.4
launchpadlib==1.10.2
lazr.restfulclient==0.13.3
lazr.uri==1.0.3
librefdatool==0.1
logilab-common==0.62.1
lxml==3.4.0
MarkupSafe==0.23
matplotlib==1.4.2
mercurial==3.5.2
mock==1.0.1
networkx==1.9
nose==1.3.4
numpy==1.9.0
nupic==0.1.0
nustudio==1.0.1
oauth==1.0.1
ordereddict==1.1
PAM==0.4.2
pexpect==3.2
Pillow==2.3.0
Pivy==0.5.0
ply==3.4
prettytable==0.7.2
psutil==1.0.1
py==1.4.24
pyasn1==0.1.7
pyasn1-modules==0.0.5
pybonjour==1.1.1
pycollada==0.4
pycparser==2.10
pycrypto==2.6.1
pygame===1.9.1release
Pygments==2.0.1
pygobject==3.14.0
pygpgme==0.3
pygraphviz==1.3rc2
pylint==1.1.0
PyMySQL==0.6.2
PyOpenGL==3.1.0
PyOpenGL-accelerate==3.1.0
pyOpenSSL==0.14
pyparsing==2.0.3
pyproj==1.9.3
pyqtgraph==0.9.8
pyserial==2.6
Pyste==0.9.10
pytest==2.4.2
pytest-cov==1.6
pytest-xdist==1.8
python-apt==0.9.3.12
python-dateutil==2.1
python-debian==0.1.27
python-debianbts==1.11
python-libtorrent==0.16.18
pytz==2012rc0
pyudev==0.16.1
pyxdg==0.25
PyYAML==3.10
pyzmq==14.4.0
reportbug==6.6.3
reportlab==3.1.8
requests==2.9.1
roman==2.0.0
scipy==0.14.0
SecretStorage==2.1.1
service-identity==1.0.0
sexpdata==0.0.3
simplejson==3.6.5
six==1.8.0
SOAPpy==0.12.22
SQLAlchemy==1.0.12
synaptiks==0.8.1
tweepy==2.1
Twisted-Core==14.0.2
Twisted-Web==14.0.2
unittest2==0.5.1
validictory==0.9.1
vboxapi==1.0
wadllib==1.3.2
Werkzeug==0.9.6
wicd==1.7.2.4
wstools==0.4.3
wxPython==3.0.1.1
wxPython-common==2.8.12.1
XlsxWriter==0.7.7
youtube-dl==2014.8.5
zope.interface==4.1.1

Jeff-Ciesielski, 2016-03-17:
Yeah, there it is: 

> Cython==0.23
> datastore==0.3.6 <-- Causing the conflict
> DBUtils==1.1

Let me see what I can do to remedy.

Jeff-Ciesielski, 2016-03-17:
Hey @marcos-paltatech- Just pushed changes to master, give that a shot and let me know if it is resolved.

marcos-paltatech, 2016-03-17:
Its working now!

Jeff-Ciesielski, 2016-03-17:
Excellent :)

closed 2016-03-17 by Jeff-Ciesielski