Jeff-Ciesielski/Boms-Away #6: A '\n' can break the schematic

marcos-paltatech, 2016-03-17:
If you paste a manufacturer into Boms-Away, lets say "Infineon Technologies", but by accident paste
"
Infineon Technologies"

the line feed gets propagated into the schematic and eeschema tells you the file is broken. You have to manually delete the extra \n in the .sch usin a text editor.

It would be mandatory to remove all \n from the string, and it would be nice to remove trailing blank spaces as well, its quite common to select extra blank spaces while copying text.

Jeff-Ciesielski, 2016-03-17:
Good catch, fields need to be sanitized.

Jeff-Ciesielski, 2016-03-17:
https://github.com/Jeff-Ciesielski/Boms-Away/commit/1202843b909ae23cb4207dbdd693517c6948697e

Should take care of this case.  Also provides a jumping off point for future enhancement.

closed 2016-03-17 by Jeff-Ciesielski