Jeff-Ciesielski/Boms-Away #7: Cannot Paste into part information text boxes.

jameswalmsley, 2016-04-13:
At least on Linux, its not possible to paste into the text boxes for partnumbers etc.
Is there a quick way to enable this... I'm digging into this kivy thing but a bit lost.

jameswalmsley, 2016-04-13:
Sorry my correction... pasting does work but text is hidden until you scroll back in the text box.
I assume this is a kivy issue and not caused by your fine work!

Thanks for the excellent project, it is saving me a lot of time.

Jeff-Ciesielski, 2016-04-13:
Hey James.  Thanks for the bug report.  I'm actually in the process of porting the tool to the wxWidgets toolkit.  While Kivy is very pythonic and 'shiny', in retrospect it just isn't a good choice for a tool like this (which is mostly business logic /  text shuffling).  I'm hoping to have feature parity in the next day or two and will release it out into the wild.  

I'm glad you're finding the tool helpful!  I'll go ahead and close this ticket.  Let me know if you hit any other issues.

closed 2016-04-13 by Jeff-Ciesielski