Jeff-Ciesielski/Boms-Away #9: How to install on OSX

jameswalmsley, 2016-04-13:
Hi Jeff,

I spent over an hour trying to get this to run on OSX.
I assume that the wxWidgets version will be easier to use, but if thats not ready a quick howto on getting it up and running on OSX would be great.

I had no problems on ubuntu.

J

mizraith, 2016-10-11:
Been working on it myself on OSX.  I'm using homebrew as a package manager, so I had to make sure Kicad was pointing to the correct python.  Right now I'm calling the following from within the BOM manager:

/usr/local/bin/python "/Users/red/Documents/!PC_Shared/KiCad_REDS_PROJECTS/BOM_Exporters/Boms-Away/bomsaway.py" "%I" "%O"

It's opening a blank wx window without any parts.....what am I missing?

Jeff-Ciesielski, 2016-10-11:
@mizraith: This isn't a kicad bom exporter, it's a standalone application that directly reads/writes kicad schematics.

Jeff-Ciesielski, 2016-10-11:
Sidenote: I no longer have access to an OSX machine, Anyone with access to one who would be willing to write up some basic instructions would be greatly appreciated :)

mizraith, 2016-10-12:
Jeff -- THAT TOTALLY HELPED!  I was completely dumbfounded as to "what to do next".....  I suggest that your main Read Me file has a section on what to do once you get to the main screen, because that is where I was stuck....

Instructions follow that should work for a reasonably mac-savvy user.  

## Prepare your Kicad Schematic Parts
You will need to have the following fields in this order:   
1.  MFR   (manufacturer)
2. MPN   (manufacturer part number)
3. SPR    (supplier)
4. SPN   (supplier part number)

**Question: Can BOMs-Away handle additional fields, like Description, Notes**

## INSTALLING Boms-Away ON OS X
### Prerequisites  
Ability to use Terminal, or at least get yourself into and out of trouble.
### Installation
1. <recommended> Install [Homebrew](http://brew.sh)     This is a great package manager for mac.
2. `brew install python`
3. `pip install sqlalchemy`
4. `brew install wx`        (or use `pip install`.  Both seem to work)
5.  Verify your installation with:  
  1. `which python`    to make sure you are using the homebrew python
  2. `python`     launch your instance and `>>>import sqlalchemy`  and `>>>import wx` to verify that you have properly installed the required packages.
6.  `cd /directory/where/you/want/bomsaway/to/live`
7.  `git clone https://github.com/Jeff-Ciesielski/Boms-Away.git`   (I _think_ git is standard on OS X now...if not, then `brew install git`)
### Launching
1.  `cd /path/to/your/kicad_project_directory`    HINT:  on the mac you can drop a folder into the Terminal window to resolve the path...saves on typing.
2   `python /path/to/bomsaway.py`    This will open the wx GUI.  

### Using
1.  File->Open    Browse to your Kicad schematic file
2. ???   I'm still learning how to use this
  _1.   Is there a way to add fields (like Description, Notes)???  These do not come across in the BOM export._

### Exporting a BOM
1.  File -> Export BOM as CSV
2. Open it up in Excel

yater90spn, 2017-02-02:
I haven't been able to close or resize KiCAD windows since installing this due to wxwidgets conflict. Didn't realize until now but thankfully others have had the issue. Going to try the patch mentioned here.

https://bugs.launchpad.net/kicad/+bug/1490762