Jeff-Ciesielski/Boms-Away #12: Cannot Open Schematics in Other Drives

Yoda12999, 2016-05-05:
BOMs Away is not able to open other drives through it's file browser on Windows. The file browser simply does not go high enough in the file structure to show other drives, stopping at the root folder of the drive that contains BOMs Away.

Jeff-Ciesielski, 2016-05-06:
That's unfortunate... I'll push the partially complete WxPython port today.  If you have a few minutes to try opening a different drive, let me know.  Hopefully I'll have that version ready for release in the next week or two and these sorts of odd Kivy related bugs will be a non-issue.

Jeff-Ciesielski, 2016-05-06:
@Yoda12999: give the wx_ui branch a shot and let me know if you can open schematics in other drives.  I'll try to get the last few blocking things finished ASAP so I can start migrating people over to that version.

Yoda12999, 2016-05-06:
The issue is fixed in the wx_ui branch as it just opens a windows explorer file selection window. On a slightly unrelated note, I like how the wx version looks, feels, and functions a whole lot more.

closed 2016-05-06 by Yoda12999