Jeff-Ciesielski/Boms-Away #16: UTF-8 Characters Cause Problems

linuxcf, 2016-08-21:
KiCAD, at least on Linux, is fully UTF-8 compliant.  So, in my existing KiCAD schematics I use strings like '4.99Ω' and '0.1μF' as values.  Such strings seem to create problems to 'bomsaway.py', which, although it does not crash, just ignores the selection, generating an error message in the terminal, e.g., for a value of '10kΩ' (without the quotes),

KeyError: u'10k\u03a9'
Traceback (most recent call last):
File "bomsaway.py", line 231, in on_fp_list
[x for x in sorted(set(self.type_data[self.fp_list.GetStringSelection()].keys()))])

When the script encounters UTF-8 characters in any field, an error like the above is printed and the particular component is just ignored.

DavidMenting, 2017-01-03:
The error seems to relate to difference in encoding between the value returned from the selection box and the keys of the component list.

1µF is encoded in the internal list as '1\xc2\xb5F', whereas the wxWidgets ListBox returns a UTF-8 encoded string u'1\uxbF'

I've tried finding a solution to this for the past hour or so, but couldn't fix it. Maybe @Jeff-Ciesielski can jump in to help or give pointers.

Gasman2014, 2017-07-16:
This affects me too. Whilst I could parse all my libraries and replace the Ω symbols with Ohm I do quite like the space saving the symbols offer. Not sure how easy a fix this is?

Jeff-Ciesielski, 2017-07-16:
A cursory glance seems to indicate that wxWidgets stores strings as UTF-32 internally.  I'll see what I can do about marshaling things back and forth, but this might be fairly non-trivial considering how much strings are used internally.

Gasman2014, 2017-07-16:
I agree that this seems to be somewhat problematic in Python 2.7 although possibly easier in Python 3. This article seems to sum up an approach to the issues quite well https://www.azavea.com/blog/2014/03/24/solving-unicode-problems-in-python-2-7/