Jeff-Ciesielski/Boms-Away #17: enable excel type export of csv

macsimski, 2017-01-30:
My fab house is soly windowz based and they have problems importing csv files from Bombs-away. Do you think it is possible to export the cv with tabs as field delimiter, quoted text and a comma and space between the multiple parts on a row

so something like 
"R101, R102, R103"\tab"100K"\tab"sweeb:0603R"\tab"3"\tab"Multicomp"\tab" etc. 

As the source of Bombs-away does not have a copyright header with version number, i cannot include a version number, but I downloaded it today 30-01-2017

changing 
`with open(base+ext, 'w') as csvfile:
            wrt = csv.writer(csvfile)`
on row 546
to 
`with open(base+ext, 'w') as csvfile:
            wrt = csv.writer(csvfile,dialect='excel')`

did not give a change. 
Unfortunately, i'm not that fluent in python. :(

macsimski, 2017-02-08:
ok. nice one. tab separated is possible with 

`with open(base+ext, 'w') as csvfile:
            wrt = csv.writer(csvfile,dialect='excel-tab')`

not excel_tab as stated in the docs. read those docs really carefull.  semicolon between the individual parts with the same parttype is not solved, but a simple search and replace can do it.

Jeff-Ciesielski, 2017-10-28:
So I took a quick crack at this and I'm not convinced that I want to tie all these features into the main source code since it's pretty specific to your board house. 

Looking at https://github.com/Jeff-Ciesielski/Boms-Away/pull/24 (which is also pretty specific), I'm thinking what might be a better fit is an export plugin system (which hilariously is what this was intended to replace in KiCad itself), but at least these plugins could be in the form of a python module.

I'll take a crack at it and will see what I can come up with.   Out of curiosity, what board house do you use (so that I might name the export plugin)

macsimski, 2017-10-29:
Cool. if you want me tor try things, just say so. My board house is electro-watt in Hilversum, the netherlands, But i think that it is not that inportant as the dialect='excel-tab' works fine here. :-)

Jeff-Ciesielski, 2017-10-29:
Give this a go: https://github.com/Jeff-Ciesielski/Boms-Away/commit/d57404cc27145f8ee092e40de6f7e07cb9bb45c3 (Should be the tip of the master branch).

macsimski, 2017-10-29:
hmm. ImportError: No module named 'wx'

I'm looking into this as i have /usr/lib/python2.7/dist-packages/wx-3.0-gtk2 installed

Jeff-Ciesielski, 2017-10-29:
Project was recently updated to Python3