Jeff-Ciesielski/Boms-Away #18: feature request: colors on entries

macsimski, 2017-02-08:
It would be of great help to see were info is missing. I don't know if this is possible with font color. like a red font color for a footprint and component until at least manufacturer, mfc number, supplier and suppl. nr. are present.

in one glimpse you can see if all parts have been populated.

your thoughts on this please

Jeff-Ciesielski, 2017-10-28:
I like this idea.  I'll try to get it in soon.

macsimski, 2017-10-29:
cool! I still like the way this works. :-)

macsimski, 2018-02-09:
hmm. this idea is not good as there is a scenario where you end up with wrong data:  make a schematic and add all component values with boms-away. then alter the schematic by copying a resistor and give it another value in kicad. open the schematic in boms-away, finding the new component with the wrong values, copied from the original resistor. So although the part has values added, they are the wrong ones, hence the color hint is in error. I'll post a feature request @ kicad for a copy part question: "copy values fields as well?"