Jeff-Ciesielski/Boms-Away #23: Empty field values are problematic when editing a component in schematic.

EvansMike, 2017-10-12:
Having saved back a schematic, editing a component in eeschema shows some|all the bomsaway added fields empty. When closing the edit dialog eeschema complains about saving empty field values and offers to delete the fields. Empty fields have to have sum value inserted in order to close the dialog without losing data.

Arguably this is a fault with KiCad, buy maybe a default value, such as "na", should be used as default value in the fields.

macsimski, 2017-10-29:
hmm. I like the idea. or maybe just a minus sign as na could be interpreted as a 'do not populate'

Jeff-Ciesielski, 2017-10-29:
Should be fixed by https://github.com/Jeff-Ciesielski/Boms-Away/commit/edd422a3afabfb9e11ee609c0667e1ae73ae37ef

Let me know if this doesn't work for you and we can figure out something that does.

closed 2017-10-29 by Jeff-Ciesielski