Jeff-Ciesielski/Boms-Away #32: Boms Away not storing values and hanging

aaronw2, 2018-09-26:
I am running into several problems with Boms Away with KiCad 5. For one thing, many components do not appear to be stored because when I re-load a schematic, many components either aren't found when I tell it to load the data, or even worse, it hangs completely.

Jeff-Ciesielski, 2018-09-26:
I'll be honest, I haven't really been keeping up with the latest KiCad developments as my day job has switched away from dealing with hardware.  

I'll see if I can't find some time to update my local setup and debug.

aaronw2, 2018-09-26:
It may have been database corruption. I erased the database and started over and it seems to be working now.

closed 2018-09-26 by aaronw2

Jeff-Ciesielski, 2018-09-26:
Easiest bug ever ;).  Thanks for following up